cortex-lab/KiloSort #244: kilosort3 - maximum variable size allowed on the device is exceeded

GuthrieDyce, 2021-08-31:
Hello world,

**Summary:**
I have been trying to sort a recording using kilosort 3 and I keep running into two errors in particular:
One of the errors I have been having returns the message "maximum variable size allowed on the device is exceeded".
If I close everything that is open on the computer except for the instance of matlab running kilosort then I get a different error during CUDA execution: "an illegal memory access was encountered".

**Specifications/Notes:**
I am using a Neuropixel 1 probe phase3b1 with staggered recording sites.
The working directory and output directories are the same (the input file is within this directory).
I am using the default specifications for this version of kilosort.
I am also using MATLAB 2019a.
The input file is 5.8 GB in size.
The hard drive used for the input, working and output folders has 3.22 TB of free space on it.
I am using Windows 10, 64 bit, 3.7 GHz, with 64 GB of RAM.
I am using an NVIDIA GeForce RTX 2060 graphics card.
It is also worth noting that I have used this computer to run kilosort on other recordings without this problem.

**Attachments:**
I have attached screenshots of the GUI, preferences, drift map, estimated drift traces, the kilosort folder (showing, as I understand it that I am in fact using kilosort 3), and the command line associated with the errors.

**Questions:**
Is the size of the input file too large?
Are the kilosort specifications wrong?
Is using the same directory for the input file as the working and output directory causing this issue?
Is there too much jitter in the recording resulting in duplication of units?  Is this overwhelming the graphics card?
Can someone help me identify (& perhaps even fix) the problem?

Thanks so much!

Guthrie

[image]

[image]

[image]

[image]
[image]

[image]

[image]
[image]

GuthrieDyce, 2021-08-31:
**Additional notes**
Both of the errors mentioned occur during the "template matching on binary file" stage.
The original drift map and estimated drift traces are associated with the first error ("maximum variable size...")
The drift map and estimated drift traces attached to this comment are associated with the second error ("illegal memory access...")
I hope this helps someone help me.

[image]
[image]

hannahay, 2021-11-22:
Hi,
I also have the same issue with illegal memory access. I am using kilosort 2.5. Did you solve the issue?

GuthrieDyce, 2021-11-22:
Hi Hannah,

I am uncertain exactly why, but I have not been having this issue lately.  The best I can do is make a few wild guesses I'm afraid:

-Make sure the input, working and output directories for kilosort are the same.

-Close other programs.

-Consider which version of MATLAB you are using (from memory 2019 works better than 2021).

-Turn your computer off and on again.

I wish I could help more.

Good luck.

Guthrie