cortex-lab/KiloSort #260: EIG did not converge

caitlinmallory, 2022-09-02:
Hi there,

I'm setting up Kilosort2 on a new computer. I was able to sort one session without issues, but I'm frequently getting the error 'Error using gpuArray/eig EIG did not converge at index = <number>'  in learnAndSolve8b-runTemplates-svdecon-eig.  NFilt is set to 512 (have also tried 1024 or leaving blank). I'm using Matlab R2022a, Cuda 11.7, GeForce 3080, linux Ubuntu. Any ideas?

robyngreene, 2022-11-23:
I am having the exact same issue on Windows 10. Did you find a way to resolve this?

tonylien, 2023-03-25:
I had this exact problem running Kilosort2 with Matlab2022a, CUDA 11.7, and Visual Studio Community C++ 2019 on Windows10. 

The issue was resolved after I downgraded Matlab to 2021a and CUDA to 11.0 (https://www.mathworks.com/help/releases/R2021b/parallel-computing/gpu-support-by-release.html) with Visual Studio Community C++ 2019.

caitlinmallory, 2023-03-25:
Thanks so much!!